RonaldSun/VI-Stereo-DSO #1: Doesn't work in Euroc

pipigenius, 2019-04-03:
Hi Ronald,
Have you tried EuroC? 
It doesn't work at all for me. Could you tell me what's the problem?
Did I take the wrong process?
Download->cmake..->make -j8->modify the bash->run
And it drifts too much!
Looking forward to your reply.
Thanks a lot xD

RonaldSun, 2019-04-04:
@pipigenius 
Your operation is correct. I updated the code, you can try it again.
I'm sorry that the performance of the project has not been fully verified because I haven't found the best weighting parameters of IMU errors to balace photometric errors. In the bash file, imu_weight is the weight of IMU errors in optimize while imu_weight_tracker in tracker. I have adjusted the weight and it seemed to run smoothly.
You can try to change the weight in the bash file to get better performance. Please remind me if you have better parameters or ways to weight IMU errors. Thanks.

pipigenius, 2019-04-09:
Thank you for reply.
Yeah I followed your instructions and it works.
But there's is a stable drift between the result's trajectory and the ground-truth's trajectory.
So I can't really tell that which parameters are better.
Did you meet the same situation? 
Thanks a lot.

RonaldSun, 2019-04-09:
@pipigenius 
I updated the code, it aligns the groundtruth. Noted that the "FullTrajectory" is the pose in metric coordinate.

xiaozhi12345678, 2019-05-30:
@pipigenius 
I run the euroc dataset，I only modify the bash file to"./build/bin/dso_dataset \
 	  files0=/home/chenyan/文档/毕业设计/${dataname}/mav0/cam0/ \
	  files1=/home/chenyan/文档/毕业设计/${dataname}/mav0/cam1/ \
 	  calib0=/home/chenyan/文档/毕业设计/VI-Stereo-DSO-master/calib/euroc/cam0.txt \
	  calib1=/home/chenyan/文档/毕业设计/VI-Stereo-DSO-master/calib/euroc/cam1.txt \
	  T_stereo=/home/chenyan/文档/毕业设计/VI-Stereo-DSO-master/calib/euroc/T_C0C1.txt \
	  imu_info=/home/chenyan/文档/毕业设计/VI-Stereo-DSO-master/calib/euroc/IMU_info.txt \
	  groundtruth=/home/chenyan/文档/毕业设计/${dataname}/mav0/state_groundtruth_estimate0/data.csv \
	  imudata=/home/chenyan/文档/毕业设计/${dataname}/mav0/imu0/data.csv \
	  pic_timestamp=/home/chenyan/文档/毕业设计/${dataname}/mav0/cam0/data.csv \
	  pic_timestamp1=/home/chenyan/文档/毕业设计/${dataname}/mav0/cam1/data.csv \
 	  preset=0 mode=1 \
	  quiet=0 nomt=1 \
          savefile_tail=/home/chenyan/文档/毕业设计/VI-Stereo-DSO-master/nt\
     	  use_stereo=1\
	  imu_weight=6 imu_weight_tracker=0.6 stereo_weight=0.5" 
but it runs one second and comes the error "core dumped",could you please give me some advice or try it?Thank you!

JiatianWu, 2019-07-26:
Hi did you succeed to run on Euroc dataset? How does it perform on Euroc? Thank you.

pipigenius, 2019-07-27:
> Hi did you succeed to run on Euroc dataset? How does it perform on Euroc? Thank you.
Yep, I've succeeded. 
But I didn't finish the evaluation. I thought the author did it and post the result in the README, didn't he?

pipigenius, 2019-07-27:
> @pipigenius
> I run the euroc dataset，I only modify the bash file to"./build/bin/dso_dataset
> files0=/home/chenyan/文档/毕业设计/${dataname}/mav0/cam0/
> files1=/home/chenyan/文档/毕业设计/${dataname}/mav0/cam1/ [...]

Sry for the delay.
But, did you follow the latest instruction?

closed 2020-09-18 by pipigenius