RonaldSun/VI-Stereo-DSO #3: Rolling shutter camera

SKioresku, 2019-05-15:
Hi! Im trying to port this library to android, but it seems that it will be problem with rolling shutter camera. How i can fix it?

RonaldSun, 2019-05-16:
I'm afraid that it may be the problem of DSO.  DSO’s performance quickly deteriorates with added geometric noise. If you can get the time interval of rolling shutter, I think maybe you can try to modify the project pixel with estimated pose of different time.

SKioresku, 2019-05-16:
I found paper "Digital Video Stabilization and Rolling Shutter Correction using Gyroscopes". I will try to realize that algorythm and apply rolling shutter correction to image before sending it to DSO pipeline

mark-72, 2019-08-27:
A great idea. I have also found several papers on this. Did you also find an implementation in C++ or another language, I always found implementations in Matlab with m-scripts.