RonaldSun/VI-Stereo-DSO #6: core dumped

xiaozhi12345678, 2019-05-29:
Hi,I use it with dataset "MH_04_difficult",I only modify the bash file,it runs maybe one second,then core dumped.It runs to main_dso_pangolin.cpp line  "delete fullSystem"  then core dumped.Would you please try it?I am looking forward to your reply.Thank you!

RonaldSun, 2019-05-29:
@xiaozhi12345678 
Sorry I'm busy graduating recently, I'll try it after a time...I think it's the problem of stereo initialization coz left-eye image has different photometric parameter with right-eye image. You can try to start feeding image from 100th image or other.

RonaldSun, 2019-07-20:
@mrfight It shoud work now. Please use latest code.

mrfight, 2019-07-22:
@RonaldSun It works, thank you!