RonaldSun/VI-Stereo-DSO #7: Alignment with Ground Truth

RongenC, 2019-05-30:
I see that when you initialize the FirstFrame IMU (FullSystem::initFirstFrame_imu), you aligned the system with the ground truth. Does this mean that the current version of code does not able to align itself to world ground with the use of IMU data?

RonaldSun, 2019-05-31:
Yes, of course. Alignment can be achieved knowing the location of the first frame.

RongenC, 2019-05-31:
What if you do not know the first frame's location (i.e. initial true world xyz)? Gravity Direction can be retrieved directly from IMU's data right?

RonaldSun, 2019-05-31:
Actually the location of first frame is set to (0,0,0). The "T_WR_align" is used to align the groundtruth to the system. So there will be a transformation between DSO's pose and world pose.

RongenC, 2019-05-31:
Yes, so my question is if there is no groundtruth available for the first
frame, this system unable to locate it's starting location (mainly height
is relevant in this case)?

RonaldSun, 2019-05-31:
Unable to lacate it's starting location

RongenC, 2019-05-31:
Thank you for the clarification!

closed 2019-05-31 by RongenC