RonaldSun/VI-Stereo-DSO #8: Is Translation from Imu-Camera matrix  important or not?

SKioresku, 2019-06-04:
I am trying to implement automatic calibration from feature points with libraryes like
https://github.com/hovren/crisp   and 
https://github.com/AaltoVision/camera-gyro-calibration 
I found, that i cant find translation part from Imu-Camera matrix. Can i just put zeroes to T vector?

RonaldSun, 2019-06-05:
I'm afraid not. I think angular velocity is ok,  but acceleration is different due to different rotation center. If the translation or angular velocity is large, the difference may be obvious.

RonaldSun, 2019-06-05:
I'm not sure, but I don't think so. If you set the translation to zero that means you assume the camera and imu have same rotation center. Different rotation center cause different acceleration. If the translation or 
angular velocity is large, the impact may be obvious.