RonaldSun/VI-Stereo-DSO #13: Must I provide the ground truth?

wangyuanbiubiubiu, 2019-07-09:
I think this code must provide the ground truth information, may I ask how can I run without providing this information.

ACFFF, 2019-07-09:
When I was running my own data, I also encountered the same problem.

wangyuanbiubiubiu, 2019-07-09:
> When I was running my own data, I also encountered the same problem.

Yes, I found that the code aligns the position of the first frame with the ground truth, so I don't know how to remove it.

RonaldSun, 2019-07-15:
I‘ve fixed the problem

shibowing, 2019-07-18:
Hi， Thanks for your sharing the code. I have tested the current code with commit "fix bug when no ground truth". However,  I found the latest code has a bug in this part (src/FullSystem/FullSystem.cpp). 
`if(gt_path.size() > 0)
        T_WR_align = T_wc * T_BC.inverse()*gt_pose[index2].inverse();<------especially have bug on this sentence.
    else
        T_WR_align = SE3();`

Have you tested this version of code on the euroc dataset? I can run code well on the previous version. However, the current version will have a segmenat fault.  
Thanks again for sharing the code!

shibowing, 2019-07-18:
> Hi， Thanks for your sharing the code. I have tested the current code with commit "fix bug when no ground truth". However, I found the latest code has a bug in this part [...]

My bad.  I have solved the problem.  Thank you!