RonaldSun/VI-Stereo-DSO #17: Process killed for some datasets

nishpatel96, 2019-08-06:
Hi Ronald,

For some datasets, for example EuroC V1_01_easy, the program works quite well for around 2000 frames but then suddenly gets killed.

These are the errors I get before and after using gdb. The errors do not occur at any specific point and I am not sure why the program gets killed.
[image]
[image]

Do you have an idea what the problem could be in this case?

nishpatel96, 2019-08-06:
Update:
It's an "Out of Memory" error but I am not able to point out what part of the code could be leading to this

RonaldSun, 2019-08-07:
I did not encounter this problem. You'd better watch the memory usage. It may be due to insufficient memory.

nishpatel96, 2019-08-08:
Yes, it is a memory issue. What was the RAM size on the PC you used for running this program?

RonaldSun, 2019-08-08:
8GB

closed 2019-08-15 by nishpatel96