RonaldSun/VI-Stereo-DSO #20: Questions in terms of EnergyFunctional::marginalizeFrame_imu function

minghanz, 2019-09-23:
Hi, 

I read through your code and I have a question in the marginalizeFrame_imu function. 

`J_all_half` is not added to any matrix after calculation. Theoretically it should be added to `HM_change_half` like how `J_all` is processed. Maybe it should be something like: 
`HM_change_half += (J_all_half.transpose()*Weight*J_all_half);`

In current implementation `HM_change_half` is always zero matrix at the end. Is it a mistake or it should actually work this way? 

Besides, I am also not quite clear about why `HM_change`, `bM_change` need to be added to `HM_imu` and `bM_imu` (same for corresponding H and b matrix in half-graph) before marginalization. Isn't the frames to be marginalized already embedded in `HM_imu` and `bM_imu`? 

Thank you!

RonaldSun, 2019-09-25:
Yes, you can try. I set `use_Dmargin` to `False` at here 
[https://github.com/RonaldSun/VI-Stereo-DSO/blob/158cc08034a87e3dc4387674976d7cbcdc97034d/src/OptimizationBackend/EnergyFunctional.cpp#L1330-L1335](url)
You can modify it yourself.

minghanz, 2019-10-04:
So you turn off `use_Dmargin` after `M_num2>25`, but before that dynamic marginalization could still happen. In that case `HM_imu` and `bM_imu` are reset to zeros when it happens (essentially removing previous marginalization information)? Does that affect the stability of the system?  

Thank you!

RonaldSun, 2019-10-05:
It shouldn't have much impact. In my test ```M_num2``` will exceed 25 very soon.