RonaldSun/VI-Stereo-DSO #23: Can't find the  static stereo residuals in the code.

HpyZhai, 2019-11-07:
Hi, thanks for your oepn source. 
I didn't find the static stereo residuals in the Stereo_DSO of HorizonAD. I see that you modified the 
original code.  Did you add  the static stereo residuals and relative optimization process?

studennis911988, 2019-11-07:
HorizonAD version do static stereo matching in **traceStereo()**  in the scope of **setCoarseTrackingRef（）**.

HpyZhai, 2019-11-08:
hi, thank you to answer my question. I checked the function traceStereo() of setCoarseTrackingRef() again. You are right，It calculated the static stereo residuals ，however，they were just used to get the best matching 2D points for the mature points in the stereo image pair。
 I didn't express my question clearly. In the stereo-dso paper,  the optimized terms are temporal multiview residuals and static stereo residals. 
In my view， the objective function （15）should be optimized overall。But in the optimization process of the key frames ，it did not take  into account。
The code in " OptimizationBackend " file was the same as the monocular DSO. I didn't find the param "lambda"(15) in the stereo-DSO code. Maybe,It had another name.
So,I checked the ”setting.cpp“ and there were no new params different from  monocular DSO。

Is there anything wrong with me？

studennis911988, 2019-11-08:
Actually, HorizonAD didn't applied Stereo-DSO fully according to original Stereo-DSO paper.

In my opinion , they treat static and temporal stereo matching as the same weight(i.e lamda = 1), which is one of the reason that makes their performance not too well.

It's not too complicated to add lamda to static stereo matching residuals in sliding window optimization, if you are really interested in it, I can provide some hint.

studennis911988, 2019-11-08:
Actually, HorizonAD didn't applied Stereo-DSO fully according to original Stereo-DSO paper.

In my opinion , they treat static and temporal stereo matching as the same weight(i.e lamda = 1), which is one of the reason that makes their performance not too well.