RonaldSun/VI-Stereo-DSO #30: You have mixed different SE3 Jacobian types

tanjunyao7, 2020-05-18:
DSO evaluates the Jacobian of the photometric error with left Jacobian of SE3. The update step of poses is also done in the left-multiplication way. However, your jacobians of IMU residuals, which probably originates from https://github.com/jingpang/LearnVIORB, adopt right Jacobian. I just wonder why the optimization still converges.

studennis911988, 2020-05-18:
The author derived the right to left Jacobian, which is also mentioned in  [issue28](https://github.com/RonaldSun/VI-Stereo-DSO/issues/28). However, I have no idea how he derived that.

tanjunyao7, 2020-05-19:
> The author derived the right to left Jacobian, which is also mentioned in [issue28](https://github.com/RonaldSun/VI-Stereo-DSO/issues/28). However, I have no idea how he derived that.

OK, I see. I havn't read through the code before.
The deriviation may be formulated as follows:
_exp(d_xi_l) T = T exp(d_xi_r)_
_exp(d_xi_l)  = T exp(d_xi_r) T^(-1) = exp(Adj(T) d_xi_r)_
_d_xi_l  = Adj(T) d_xi_r_

studennis911988, 2020-05-20:
@tanjunyao7 Your derivation really make sense, thanks a lot!

DengueTim, 2024-09-15:
But optimisation of the transform from the IMU World frame to DSO's frame uses the right Jacobian right?   I'm struggling to understand the intuition of eq. A.6 in Stumberg's paper's supplement...