RonaldSun/VI-Stereo-DSO #32: E=λE_stereo+E_imu

Ginger-Light, 2020-12-28:
Hi,

Thanks for sharing the code.

I have the issue, I try  to find the λ factor before E_stereo in the code, In detail, it shoulb be in 
`void EnergyFunctional::solveSystemF()`
but I didn't find it.  

Excuse me, it was not added λ here, or I was not carefulty enough to find it.

Thanks again for your code

studennis911988, 2020-12-28:
The λ factor in the objective function actually goes into the information matrix(W) in G-N normal equation (J^TWJ x = -JWr), so you can find the  λ factor in `double PointFrameResidual::linearizeStereo(CalibHessian* HCalib)` which is located in Residuals.cpp.

`float lambda = stereo_weight;`

Wish this might help! :)

Ginger-Light, 2020-12-28:
> The λ factor in the objective function actually goes into the information matrix(W) in G-N normal equation (J^TWJ x = -JWr), so you can find the λ factor in `double [...]

My friend! Thank u very much! 

This is Lagrange Multiplier, right?

studennis911988, 2020-12-28:
It has nothing to do with Lagrange Multiplier.
Since DSO is a direct method, it can't normalized the norm of visual and imu residual to the non-dimensional terms（ Mahalanobis distance）as it's typically done in featured-based VIO system, so VI-DSO just uses a λ to weight between visual and imu energy function.

Ginger-Light, 2020-12-28:
> It has nothing to do with Lagrange Multiplier.
> Since DSO is a direct method, it can't normalized the norm of visual and imu residual to the non-dimensional terms（ Mahalanobis distance）as it's [...]

Thank you for helping me understand this！